zm0612/funny_lidar_slam #4: 保存地图报错

mobi1019, 2024-09-19:
您好，我在保存地图的时候执行命令后报错，您看是什么原因呢？
`robot@robot:~/funny_lidar_slam_ws$ rosservice call /funny_lidar_slam/save_map "map_path: '/home/robot/funny_lidar_slam_ws/map' split_map: false"
Traceback (most recent call last):
  File "/opt/ros/noetic/bin/rosservice", line 35, in <module>
    rosservice.rosservicemain()
  File "/opt/ros/noetic/lib/python3/dist-packages/rosservice/__init__.py", line 767, in rosservicemain
    _rosservice_cmd_call(argv)
  File "/opt/ros/noetic/lib/python3/dist-packages/rosservice/__init__.py", line 628, in _rosservice_cmd_call
    service_args.append(yaml.safe_load(arg))
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 162, in safe_load
    return load(stream, SafeLoader)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 438, in parse_block_mapping_key
    raise ParserError("while parsing a block mapping", self.marks[-1],
yaml.parser.ParserError: while parsing a block mapping
  in "<unicode string>", line 1, column 1:
    map_path: '/home/robot/funny_lid ... 
    ^
expected <block end>, but found '<scalar>'
  in "<unicode string>", line 1, column 49:
     ... /robot/funny_lidar_slam_ws/map' split_map: false`

mobi1019, 2024-09-19:
robot@robot:~/funny_lidar_slam_ws$ rosservice call /funny_lidar_slam/save_map "map_path: '/home/robot/funny_lidar_slam_ws/map' split_map: false"
Traceback (most recent call last):
  File "/opt/ros/noetic/bin/rosservice", line 35, in <module>
    rosservice.rosservicemain()
  File "/opt/ros/noetic/lib/python3/dist-packages/rosservice/__init__.py", line 767, in rosservicemain
    _rosservice_cmd_call(argv)
  File "/opt/ros/noetic/lib/python3/dist-packages/rosservice/__init__.py", line 628, in _rosservice_cmd_call
    service_args.append(yaml.safe_load(arg))
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 162, in safe_load
    return load(stream, SafeLoader)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 438, in parse_block_mapping_key
    raise ParserError("while parsing a block mapping", self.marks[-1],
yaml.parser.ParserError: while parsing a block mapping
  in "<unicode string>", line 1, column 1:
    map_path: '/home/robot/funny_lid ... 
    ^
expected <block end>, but found '<scalar>'
  in "<unicode string>", line 1, column 49:
     ... /robot/funny_lidar_slam_ws/map' split_map: false

zm0612, 2024-09-19:
你好，保存地图时，如果使用了自定义路径，需要给完整的路径和地图名称，例如,
`rosservice call /funny_lidar_slam/save_map "map_path: '/home/xx/xx/map.pcd' split_map: false"`

另外还要确保，路径对应的文件夹存在。

85256638, 2024-11-15:
> 你好，保存地图时，如果使用了自定义路径，需要给完整的路径和地图名称，例如, `rosservice call /funny_lidar_slam/save_map "map_path: '/home/xx/xx/map.pcd' split_map: false"`
> 
> 另外还要确保，路径对应的文件夹存在。

找到问题了，应该加{}

`rosservice call /funny_lidar_slam/save_map "{map_path: '/home/xx/xx/map.pcd', split_map: false}"`