zm0612/funny_lidar_slam #5: 编译报错

zhlx007, 2024-09-26:
作者，你的pcl版本是多少？我编译的时候报错：
/home/test/navigation/funny_lidar_slam_ws/src/funny_lidar_slam/src/slam/preprocessing.cpp:475:23: error: ‘isFinite’ is not a member of ‘pcl’; did you mean ‘isValueFinite’?
  475 |             if (!pcl::isFinite(cloud_none[i])) {
      |                       ^~~~~~~~
      |                       isValueFinite
funny_lidar_slam/CMakeFiles/funny_lidar_slam_lib.dir/build.make:369: recipe for target 'funny_lidar_slam/CMakeFiles/funny_lidar_slam_lib.dir/src/slam/preprocessing.cpp.o' failed
make[2]: *** [funny_lidar_slam/CMakeFiles/funny_lidar_slam_lib.dir/src/slam/preprocessing.cpp.o] Error 1
make[2]: *** 正在等待未完成的任务....
CMakeFiles/Makefile2:2921: recipe for target 'funny_lidar_slam/CMakeFiles/funny_lidar_slam_lib.dir/all' failed
make[1]: *** [funny_lidar_slam/CMakeFiles/funny_lidar_slam_lib.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j16 -l16" failed

zm0612, 2024-09-26:
hello，我使用的是ROS Noetic自带的pcl，如果我没有记错的话，应该是pcl-1.10

zm0612, 2024-09-26:
如果你的本地环境一直没法编译通过，可以尝试使用docker，我提供的有docker环境

zm0612, 2024-09-26:
我觉得这里倒不是代码BUG，更多的原因还是因为你的环境存在部匹配，尽量还是使用ros noetic和ubuntu20。如果环境实在没法满足，尝试使用docker。