zm0612/funny_lidar_slam #6: MID360建图

RightTr, 2024-09-27:
作者你好，我想用MID360雷达直接开始建图，运行launch文件后，rviz里没有建图结果，/livox/lidar也并没有消息。当我运行了livox driver的msg启动文件后，也并没有建图效果。
[image]

zm0612, 2024-09-27:
你好，利用MID360实时建图的话，可以按照如下思路进行参考：
1. 首先你需要有一台电脑，轻便的笔记本电脑为宜，方便你自由移动；
2. 确保你的MID360与电脑主机物理连接通畅，并且你需要给MID360供电，如果需要大范围移动，则需要移动电源；
3. 然后在你的电脑上配置环境，按照我的代码仓库的README第2节去操作。其中你需要额外去安装MID360的ros驱动，这一步README中没有，可以通过livox的官网去找文档完成。
4. 如果环境安装没有问题，并且激光雷达也能正常驱动，那你可以按照README第3节，编译funny-lidar-slam程序包，然后按照第4节去运行建图程序。由于是实时建图，不是播放bag包，因此你还需要通过livox提供的ros驱动包，将mid360的launch文件给运行起来，以保证它能实时输出点云和imu数据。
5. 以上就是整个流程了，可以去尝试一下

zm0612, 2024-09-27:
需要注意，mid-360的ros驱动，输出的点云有多种格式.

- 如果你选择的是`sensor_msgs/PointCloud2`，则运行funny-lidar-slam的下述建图命令
`roslaunch funny_lidar_slam mapping_mid360.launch`

- 如果你选择的是`livox_ros_driver2/CustomMsg`，则运行funny-lidar-slam的下述建图命令
`roslaunch funny_lidar_slam mapping_livox_avia.launch`

wang119YN, 2025-02-25:
[image]
我使用mid360雷达进行实时建图，并且输出的点云格式为livox_ros_driver2/CustomMsg，但是运行建图命令后没有点云出现，也没有建图成功，包括点云输出格式为sensor_msgs/PointCloud2，运行roslaunch funny_lidar_slam mapping_mid360.launch此建图命令也没有点云显示，请问知道什么原因吗？

lwcode, 2025-08-16:
> [image] 我使用mid360雷达进行实时建图，并且输出的点云格式为livox_ros_driver2/CustomMsg，但是运行建图命令后没有点云出现，也没有建图成功，包括点云输出格式为sensor_msgs/PointCloud2，运行roslaunch funny_lidar_slam mapping_mid360.launch此建图命令也没有点云显示，请问知道什么原因吗？

我也遇到了一摸一样的问题！！！！！