zm0612/funny_lidar_slam #15: 关于mid360在开启回环检测的情况下建图失败

Phymin, 2024-11-25:
作者，你好，我在自己采集的mid360数据集上建图，如果在开启回环检测的情况下会出现建图失败的问题。能否帮忙看下是什么问题，谢谢。我的数据集链接：[数据集](https://pan.baidu.com/s/1OosvK7ChqnyAtDTJTP-DnA?pwd=r8dp 提取码: r8dp)，建图时使用的配置：
```
sensor_topic:
        lidar_topic: "/livox/lidar"
        imu_topic: "/livox/imu"

    slam_mode: 1  # 1: Mapping, 2: Localization

    lidar:
        # Velodyne_16, Velodyne_32, Velodyne_64, Ouster_128_os1,
        # LeiShen_16, RoboSense_16, Livox_Avia, Livox_Mid_360, None
        lidar_sensor_type: Livox_Mid_360
        lidar_point_jump_span: 3
        lidar_point_time_scale: 1.0e-9
        lidar_use_min_distance: 0.5
        lidar_use_max_distance: 1000.0
        lidar_rotation_noise_std: 0.001
        lidar_position_noise_std: 0.01

    imu:
        init_acc_bias: 0.0
        init_gyro_bias: 0.0
        acc_noise_std: 0.1
        gyro_noise_std: 0.1
        acc_rw_noise_std: 1.0e-4
        gyro_rw_noise_std: 1.0e-4
        has_orientation: false
        data_searcher_buffer_size: 5000

    gravity: 9.81

    calibration:
        lidar_to_imu: [ 1.0, 0.0, 0.0, -0.011,
                        0.0, 1.0, 0.0, -0.02329,
                        0.0, 0.0, 1.0, 0.04412,
                        0.0, 0.0, 0.0, 1.0 ]

    frontend:
        fusion_opti_iters: 30
        fusion_method: TightCouplingOptimization # LooseCoupling, TightCouplingKF, TightCouplingOptimization

        registration_and_searcher_mode: PointToPlane_IVOX
        feature:
            planar_voxel_filter_size: 0.2  #default: 0.5
        registration:
            optimization_iter_num: 30
            point_to_planar_thres: 0.1
            position_converge_thres: 0.001
            rotation_converge_thres: 0.005

    system:
        keyframe_delta_distance: 0.5
        keyframe_delta_rotation: 0.2
        enable_loopclosure: true
        enable_visualize_global_map: true
        global_map_visualization_resolution: 0.5
        tile_map_grid_size: 100.0

    loopclosure:
        registration_converge_threshold: 1.5
        skip_near_loopclosure_threshold: 100
        skip_near_keyframe_threshold: 100
        candidate_local_map_left_range: 20
        candidate_local_map_right_range: 20
        loopclosure_local_map_left_range: 30
        near_neighbor_distance_threshold: 10.0
```

zm0612, 2024-11-26:
hello, 你发过来的数据，我进行了测试。
- 由于你使用的是激光雷达数据格式为：livox_ros_driver2/CustomMsg，所以我直接运行了这个launch文件：launch/mapping_livox_avia.launch
- 并修改了config文件：config/mapping/config_livox_avia.yaml第52行的参数 enable_loopclosure: true

运行效果如下：
[image]
[image]
[image]

通过观察，未见墙面分层，以及轨迹错乱，也未发现有回环错误的情况。

不过观察终端中输出的log，有如下警告：
[image]
这个并不是错误，这是为了方便查看回环检测是否成功而打出来的log。由于回环检测是实时在进行的，即使终端中出现了回环检测失败的警告，也会有部分是成功的，回环检测成功的log并没有打印出来，如果你需要可以尝试在如下位置打印log：https://github.com/zm0612/funny_lidar_slam/blob/611cc5893933af06716f254fa21728b977d977bf/src/slam/loop_closure.cpp#L103

另外，说句题外话，分享一下我对回环检测的一些经验，实际上激光雷达的回环检测是比较脆弱的，多数情况下是依赖几何结构特征的，因此在走廊的情形下容易错误跨楼层或者前后大范围错误的回环结果。特别是我这里采用的基于欧氏距离的方式找候选帧。多数情况要解决这种问题，一方面你要调整回环的参数，缩小候选帧的搜索距离，增加更鲁邦的判断条件，比较说做一些退化检测等等，去进一步提升回环的可靠性，另外对于工程应用来说，可以人为去设计一些行进路线，使得轨迹交叠在结构复杂的地方，这样回环更容易成功。

希望对你有帮助，如果有问题，欢迎随时提出来。

Phymin, 2024-11-27:
不好意思，你确定是使用我分享的数据包吗？我的数据包是point cloud格式的点云数据，可以查看下图：
[image]

不开启回环检测的建图效果：
[image]
开启回环检测的建图效果：
[image]

zm0612, 2024-11-27:
哎呀😅，实在抱歉，昨天晚上时间太匆忙，我应该是把你和另外一个朋友的数据弄混了，今天晚上回家我重新给你看，实在不好意思

zm0612, 2024-11-27:
hello, 我这次应该是数据下载对了，哈哈，但是看起来你传给我的是ros2的数据，格式为db3，如下图
[image]

我尝试转成了ros1的bag文件，运行funny lidar slam时程序出现了coredump，排查发现你给我的数据包存在imu和lidar掉帧的问题（如下图）
[image]

麻烦再帮我核对一下  是哪里出问题了。
@Phymin

Phymin, 2024-11-28:
Hello, 非常感谢你的回复，不好意思传错了数据包，不过你说的coredump问题我也遇到了，我没注意到是掉帧问题，所以我修改了一下你的源代码，就把这个问题跳过去了，修改的源码如下：
```
if (left_boundary_iter != right_boundary_iter)
{
    while (--left_boundary_iter != right_boundary_iter)
    {
        data_segment.push_back(*left_boundary_iter);
   }
}
```
我重新传了个ros1的数据包，不过这个掉帧问题依然存在的，我是直接录制的mid360的原始数据，这个掉帧问题以你的经验，有可能是怎么出现的呢？我会尝试更新下驱动，看看会不会解决这个问题。另外这个掉帧问题是造成回环建图失败的原因吗？谢谢。
[数据集](https://pan.baidu.com/s/1zy1jImZeTJGLJ_YoqTrXPg?pwd=rkuj) 提取码: rkuj

zm0612, 2024-11-28:
hello，新数据我测试了，和之前是一样的问题，有掉帧问题，这个导致整个数据包无法完成一次成功的建图。并且IMU的数据也存在异常。我觉得你首先需要去解决数据掉帧的问题，确保数据正常之后，再去考虑回环检测的问题。

对于掉帧的问题，我有几个思路你可以按如下步骤顺序递进的去查找：
- 1 首先确认是否是主机cpu不够用，可以使用Ubuntu终端htop命令去实时查看CPU的占有率。如果确实存在cpu异常高的占有率，则考虑关闭无关程序，或者换更强的电脑试一试。
- 2 确实驱动程序是不是livox原版的，不要带有任何改动。一般来说官方的ros驱动程序，正确性是有保证的。
- 3 确认主机网口的速率不要太低，并且网线与主机要稳定连接，如果中途出现断联，录的ros包会存在掉帧。当然也要确认激光雷达硬件不存在故障，或者过热的问题。如果有多余激光雷达换一个试试。
- 4 如果以上都做了还是找不到原因，那可以尝试更底层的检查方法，这样无疑也会更复杂，直接录取网线的pcap数据，看激光雷达原始发出来的数据是否完整。

另外，我不建议修改代码去适配有问题的数据包，这样会导致一个问题引发一连串的问题，最终让你的排查陷入两难。

如果数据掉帧问题解决了，再考虑回环错误的问题，对于我的代码如果实在不能在你的环境中成功回环，需要你去尝试调一些参数，主要就是以下这些会影响回环效果：
    loopclosure:
        registration_converge_threshold: 1.5  # 回环配准时候的分数值，分数越小召回率月底，精度就会越高
        skip_near_loopclosure_threshold: 100  # 生成多少个keyframe之后再开始进行回环检测
        skip_near_keyframe_threshold: 100  # 回环一次之后，跳过多少个关键帧再执行回环检测
        candidate_local_map_left_range: 20 # 以回环候选帧往前拼接多少关键帧点云
        candidate_local_map_right_range: 20 # 以回环候选帧往后拼接多少关键帧点云
        loopclosure_local_map_left_range: 30 # 以回环帧往前拼接多少关键帧点云
        near_neighbor_distance_threshold: 10.0 # 以当前回环帧为半径搜索距离自己最近的帧作为回环候选帧

其中，对于小环境，可以首先尝试调小registration_converge_threshold,near_neighbor_distance_threshold，来做尝试。

希望对你有帮助 @Phymin

Phymin, 2024-11-28:
好的，我先尝试解决掉帧的问题，然后再看回环的问题，如果还有问题，再来请教。非常感谢你的耐心解答，对我非常有帮助。

Phymin, 2024-11-30:
Hello, 我更新了mid360的驱动，然后重新录制了一个[数据包](https://pan.baidu.com/s/1lmgT9bEsvd_59a06fAGQ7g?pwd=de1m )，没出现coredump，应该没有丢帧的问题了。按照你的指导，更改了回环参数，但是在检测到回环后，地图就乱了，跟之前差不多的现象，能否帮忙看看是哪里的问题，谢谢。 @zm0612 
我使用的参数如下：
```
    sensor_topic:
        lidar_topic: "/livox/lidar"
        imu_topic: "/livox/imu"

    slam_mode: 1  # 1: Mapping, 2: Localization

    lidar:
        # Velodyne_16, Velodyne_32, Velodyne_64, Ouster_128_os1,
        # LeiShen_16, RoboSense_16, Livox_Avia, Livox_Mid_360, None
        lidar_sensor_type: Livox_Mid_360
        lidar_point_jump_span: 1
        lidar_point_time_scale: 1.0e-9
        lidar_use_min_distance: 0.5
        lidar_use_max_distance: 1000.0
        lidar_rotation_noise_std: 0.001
        lidar_position_noise_std: 0.01

    imu:
        init_acc_bias: 0.0
        init_gyro_bias: 0.0
        acc_noise_std: 0.1
        gyro_noise_std: 0.1
        acc_rw_noise_std: 1.0e-4
        gyro_rw_noise_std: 1.0e-4
        has_orientation: false
        data_searcher_buffer_size: 5000

    gravity: 9.81

    calibration:
        lidar_to_imu: [ 1.0, 0.0, 0.0, -0.011,
                        0.0, 1.0, 0.0, -0.02329,
                        0.0, 0.0, 1.0, 0.04412,
                        0.0, 0.0, 0.0, 1.0 ]

    frontend:
        fusion_opti_iters: 30
        fusion_method: TightCouplingOptimization # LooseCoupling, TightCouplingKF, TightCouplingOptimization

        registration_and_searcher_mode: PointToPlane_IVOX
        feature:
            planar_voxel_filter_size: 0.2
        registration:
            optimization_iter_num: 30
            point_to_planar_thres: 0.1
            position_converge_thres: 0.001
            rotation_converge_thres: 0.005

    system:
        keyframe_delta_distance: 0.5
        keyframe_delta_rotation: 0.1
        enable_loopclosure: true
        enable_visualize_global_map: true
        global_map_visualization_resolution: 0.5
        tile_map_grid_size: 100.0

    loopclosure:
        registration_converge_threshold: 0.5
        skip_near_loopclosure_threshold: 50
        skip_near_keyframe_threshold: 50
        candidate_local_map_left_range: 20
        candidate_local_map_right_range: 20
        loopclosure_local_map_left_range: 30
        near_neighbor_distance_threshold: 3.0

```

Phymin, 2024-12-01:
好的，大概明白了，非常感谢

closed 2024-12-01 by Phymin