zm0612/funny_lidar_slam #16: 关于mid360定位波动比较大的问题

Phymin, 2024-12-06:
作者你好，我现在使用mid360来做移动小车的定位，总体上来说，定位没有什么问题，但是精度上看起来不是很好，会存在一些波动，特别是小车从运行状态到停止状态，有时候能明显看到在小车停止之后，定位发生了跳动，定位使用的算法是IcpOptimized，想请教一下，有没有什么办法来解决或缓解这一问题呢？如果你需要更多的信息，也请说明，我再提供，谢谢。
```
sensor_topic:
        lidar_topic: "/livox/lidar"
        imu_topic: "/livox/imu"

    slam_mode: 2  # 1: Mapping, 2: Localization

    lidar:
        # Velodyne_16, Velodyne_32, Velodyne_64, Ouster_128_os1,
        # LeiShen_16, RoboSense_16, Livox_Avia, Livox_Mid_360, None
        lidar_sensor_type: Livox_Mid_360
        lidar_point_jump_span: 3
        lidar_point_time_scale: 1.0e-9
        lidar_use_min_distance: 1.5
        lidar_use_max_distance: 100.0
        lidar_rotation_noise_std: 0.001
        lidar_position_noise_std: 0.01

    imu:
        init_acc_bias: 0.0
        init_gyro_bias: 0.0
        acc_noise_std: 0.1
        gyro_noise_std: 0.1
        acc_rw_noise_std: 1.0e-4
        gyro_rw_noise_std: 1.0e-4
        has_orientation: false
        data_searcher_buffer_size: 5000

    gravity: 9.81

    calibration:
        lidar_to_imu: [ 1.0, 0.0, 0.0, -0.011,
                        0.0, 1.0, 0.0, -0.02329,
                        0.0, 0.0, 1.0, 0.04412,
                        0.0, 0.0, 0.0, 1.0 ]

    frontend:
        fusion_opti_iters: 30
        fusion_method: TightCouplingOptimization # LooseCoupling, TightCouplingKF, TightCouplingOptimization

        registration_and_searcher_mode: IcpOptimized
        registration:
            local_map_size: 0
            point_search_thres: 1.0
            optimization_iter_num: 30
            local_map_cloud_filter_size: 0.3  # default: 0.4
            source_cloud_filter_size: 0.3     # default: 0.4
            position_converge_thres: 0.005
            rotation_converge_thres: 0.005
            keyframe_delta_distance: 0.0
            keyframe_delta_rotation: 0.0
```

zm0612, 2024-12-06:
你好，定位波动大概在什么范围啊？可以通过rviz中的grid功能，大致估计一下。定位本身的效果对建图质量和激光雷达检测的稳定性有较强的依赖，可以尝试建图时运行机器人更加平稳，避免大的加速和减速，另外转弯时尽量慢一些，避免原地快速的自转，多尝试尝试尽量让建图质量更好。

另外，如果对定位的稳定性要求较高，可以尝试减小imu相关参数的方差，将imu在定位结果中的权重加大，这样可以保证轨迹局部更加平滑。可以尝试调整的参数已在下述中描述。
```cpp
imu:
        init_acc_bias: 0.0
        init_gyro_bias: 0.0
        acc_noise_std: 0.1  # 尝试调小这个参数
        gyro_noise_std: 0.1 # 尝试调小这个参数
        acc_rw_noise_std: 1.0e-4
        gyro_rw_noise_std: 1.0e-4
        has_orientation: false
        data_searcher_buffer_size: 5000
```

还可以尝试减少激光雷达的检测距离，可修改参数如下：
```cpp
    lidar:
        # Velodyne_16, Velodyne_32, Velodyne_64, Ouster_128_os1,
        # LeiShen_16, RoboSense_16, Livox_Avia, Livox_Mid_360, None
        lidar_sensor_type: Livox_Mid_360
        lidar_point_jump_span: 3
        lidar_point_time_scale: 1.0e-9
        lidar_use_min_distance: 1.5 # 这个值可尝试修改
        lidar_use_max_distance: 100.0 # 这个值可以尝试修改，改小一些，提升会比较明显
        lidar_rotation_noise_std: 0.001
        lidar_position_noise_std: 0.01
```

希望对你有帮助，如果需要我帮你一起调试，可以分享一份你的地图和定位的bag包给我。

Phymin, 2024-12-07:
非常感谢你的回复，我目前发现的波动能达到100mm左右，建图的稳定性我觉得应该没问题，因为速度很慢，直行速度在0.3m/s, 旋转速度在6度/s。 我先调参试试效果，再来请教。

Phymin, 2024-12-09:
你好，我现在把imu的noise减小到0.01，也把激光头最远距离调成20.0，从测试的情况来看，波动还是比较大，小车运行起来控制精度也比较差。我把数据集和地图放到[链接](https://pan.baidu.com/s/1l801YGD-6dgdx841qTzRFQ?pwd=6am5)，其中数据包中也包含了另外一个第三方的定位数据，第三方的定位数据是通过一个2D雷达得到，因为它的精度很高，所以可以作为参考。mid360和这个2D雷达的安装位置差距不大，除了z轴之外。因为两份地图的原点无法完全对齐，所以输出的定位数据会有差异，主要看波动情况就好。 还请看下是否能进一步降低mid360的定位波动， 谢谢 @zm0612

zm0612, 2024-12-09:
我尝试修改了一些参数(下述已经注明)，看上去效果是有提升的，但是波动有明显减小，但是也还是存在一些波动的。
```cpp
lidar:
    # Velodyne_16, Velodyne_32, Velodyne_64, Ouster_128_os1,
    # LeiShen_16, RoboSense_16, Livox_Avia, Livox_Mid_360, None
    lidar_sensor_type: Livox_Mid_360
    lidar_point_jump_span: 3
    lidar_point_time_scale: 1.0e-9
    lidar_use_min_distance: 0.5
    lidar_use_max_distance: 40.0 # 改小了
    lidar_rotation_noise_std: 0.001
    lidar_position_noise_std: 0.01

imu:
    init_acc_bias: 0.0
    init_gyro_bias: 0.0
    acc_noise_std: 0.01 # 改小了
    gyro_noise_std: 0.03 # 改小了
    acc_rw_noise_std: 1.0e-4
    gyro_rw_noise_std: 1.0e-4
    has_orientation: false
    data_searcher_buffer_size: 5000

gravity: 9.81

calibration:
    lidar_to_imu: [ 1.0, 0.0, 0.0, -0.011,
                    0.0, 1.0, 0.0, -0.02329,
                    0.0, 0.0, 1.0, 0.04412,
                    0.0, 0.0, 0.0, 1.0 ]

frontend:
    fusion_opti_iters: 30
    fusion_method: TightCouplingOptimization # LooseCoupling, TightCouplingKF, TightCouplingOptimization

    registration_and_searcher_mode: IcpOptimized
    registration:
        local_map_size: 0
        point_search_thres: 0.5 #改小了
        optimization_iter_num: 30
        local_map_cloud_filter_size: 0.3 # 改小了
        source_cloud_filter_size: 0.2 # 改小了
        position_converge_thres: 0.005
        rotation_converge_thres: 0.005
        keyframe_delta_distance: 0.0
        keyframe_delta_rotation: 0.0
```
[image]
- 上图一个栅格的长度是5厘米，机器人静止时定位结果有波动，但是波动基本在2厘米以内。直线行驶时轨迹整体是比较顺滑的，没有明显的波动和跳变（如下图的直线行驶轨迹）。

[image]

[image]
- 有比较明显跳动的地方时是机器人转弯时，上述转弯的地方在参数没有微调时跳动较大，调整之后有明显减少，但是依然还是有跳动。我觉得有这么几方面的原因：（1）代码中定位算法使用的是icp，icp算法在配准时source点云和target点云不能差异太大，而转弯时视角变化较快，确实对icp是一种挑战。由于时间原因，我没有再细致的去调整参数，我有一个思路你可以去尝试一下，把你建图时候的点云map，降采样的size改小一些到0.1试一下。（2）另外定位的时候机器人转弯再慢一些观察定位结果会不会更好，根据我之前经验，定位和控制会耦合在一起使得系统发散，更通俗的讲，定位如果有跳动，然后控制也会跳动，两个跳动相互影响会使得定位结果跳动加剧，所以我建议你尝试转弯再慢一些看看是不是定位效果就会更好。（3）如果确实减慢转弯速度和提升地图的稠密度，定位效果有改善，但是还是有无法接受的跳动，这个时候我认为纯靠调参数已经比较难解决了，因为建图上的误差已经成为了系统误差消除不掉了，而激光雷达自身也会因为材质和反射率的不同存在随机噪声误差，此时我建议尝试进行滤波或者多传感器融合，最简单的可以先快速尝试低通滤波消除轨迹短时间的跳变，看最终的整体机器人效果如何。但是也不要忘记了，低通滤波存在滞后，当你机器人速度加快时定位可能跟不上，此时如果你机器人有轮速的里程计，可以尝试将轮速和激光雷达的定位结果进行融合，应该是会得到比较好的定位结果的，可以保证全局是一致的，局部是平滑的。
可以按照上述的思路去尝试一下。 @Phymin 

说一些稍微题外一些的话，根据我过往的经验，对于室内机器人，如果应用在不是特别宽阔的环境中，实际上多线激光雷达的定位效果不会明显优于单线激光雷达，甚至有时候效果还不如单线激光雷达的定位效果。多线激光雷达的优势在于观测范围更广，所以定位的鲁棒性更强，但是由于观测数据过多有时候观测与地图存在差异，过多的点云数据会导致误差加剧。反倒是单线激光雷达的检测数据更干净，定位效果反而会更好。对于室内场景，使用单线激光雷达构建栅格地图，然后定位时使用粒子滤波，最后再将定位结果与轮式里程计进行融合，这样一套流程，实际上多数情况下也能有不错的效果。

Phymin, 2024-12-10:
非常感谢你这么详细的回复，关于你说的降采样size改为0.1，是指保存全局地图的降采样吗？如果是的话，我现在发给你的地图就是0.1的。
我决定使用mid360来做室内场景的建图和定位，也是因为看到了你这份开源代码，虽然之前也看了一些其他的开源代码，但都没有你的全面和清晰。所以我的初衷是想使用3D slam来解决室内和室外的建图和定位。听你这么一说，这个方案好像也不是完全行得通，对于单线激光雷达的开源方案，你有没有推荐的？你有尝试过slamtoolbox吗？谢谢。

zm0612, 2024-12-10:
对于3D激光雷达用在室内定位的情况，也可以尝试一下替换掉icp配准，使用点到面的loam去试一下，用以下参数去替换：
https://github.com/zm0612/funny_lidar_slam/blob/c103760769ee2c39d91f55c1a89f7401ca65343a/config/localization/config_turing.yaml#L42-L53

其中可以尝试修改如下参数
```cpp
         optimization_iter_num: 10  # 可以适当加大
         point_to_planar_thres: 0.1 # 可以适当加大，每次增加0.1
         position_converge_thres: 0.01  # 根据实际效果适当缩小这个参数
         rotation_converge_thres: 0.01  # 根据实际效果适当缩小这个参数
```

由于我之前在室内使用3D激光定位较少，没有花太多时间去调试室内的场景。但是我推测使用上述点到面的LOAM效果应该会有提升，但是应该也需要调一调参数，甚至要改一些代码。如果有时间你可以去尝试尝试。后续我闲一点儿了我也来调一下。

总得来说如果你想提升定位效果，但是地图质量又没法提升，尝试使用更好的点云配准和使用多传感器融合是两个方向。

对于单线激光雷达建图比较成熟了，我推荐使用google的cartographer进行建图，定位的话使用ROS的AMCL功能包，结合这两个包，室内应该也是能用的，但是需要调试的参数不少，另外也对轮式里程计有较大的依赖

Phymin, 2024-12-10:
好的，我多试试，再次感谢

Phymin, 2024-12-10:
> 对于3D激光雷达用在室内定位的情况，也可以尝试一下替换掉icp配准，使用点到面的loam去试一下，用以下参数去替换：
> 
> https://github.com/zm0612/funny_lidar_slam/blob/c103760769ee2c39d91f55c1a89f7401ca65343a/config/localization/config_turing.yaml#L42-L53
> 
> 其中可以尝试修改如下参数
> 
> ```c++
>          optimization_iter_num: 10  # 可以适当加大
>          point_to_planar_thres: 0.1 # 可以适当加大，每次增加0.1
>          position_converge_thres: 0.01  # 根据实际效果适当缩小这个参数
>          rotation_converge_thres: 0.01  # 根据实际效果适当缩小这个参数
> ```
> 
> 由于我之前在室内使用3D激光定位较少，没有花太多时间去调试室内的场景。但是我推测使用上述点到面的LOAM效果应该会有提升，但是应该也需要调一调参数，甚至要改一些代码。如果有时间你可以去尝试尝试。后续我闲一点儿了我也来调一下。
> 
> 总得来说如果你想提升定位效果，但是地图质量又没法提升，尝试使用更好的点云配准和使用多传感器融合是两个方向。
> 
> 对于单线激光雷达建图比较成熟了，我推荐使用google的cartographer进行建图，定位的话使用ROS的AMCL功能包，结合这两个包，室内应该也是能用的，但是需要调试的参数不少，另外也对轮式里程计有较大的依赖

我使用了这个方法，发现定位波动的改善很明显，然后现在控制精度大部分情况也能在±10mm之内，但最早的提出的那个问题有时候还是会出现：小车停下来之后，定位发生比较大的跳变，刚看到一次有差不多30mm，这种问题有办法改善吗？另外我有使用EKF融合小车里程计+IMU（只使用了Z轴角速度）+激光定位数据，不过对于这种跳变好像没法缓解，也有可能我写的EKF有问题，因为之前第三方的定位精度很高，所以没能发现问题，不知道你有没有推荐的融合实现，谢谢。

zm0612, 2024-12-10:
定位跳30mm的话，可能还是得具体分析是什么原因造成的。去看一下是机器人平台不够稳定，运动太剧烈，点云突然出现了遮挡，总之原因较多需要拿数据去具体分析了。

对于融合方法的话，主要就是滤波和优化。

滤波的话基本上都是选择的EKF，这个实现起来还是比较简单的，其中对于观测你需要动态的调整观测方程的方差（也就是激光雷达定位结果的方差），这一点我的激光雷达定位输出时并没有对定位结果输出方差或者置信度，但是这个是必须的，你需要去增加这个功能。可以使用配准之后的情况来反应当前定位结果的置信度，比方说按照fitnessscore，我的配准算法基本都有这个计算的实现方法，然后增加一个系数去统一预测和观测之间方差的单位，这个系数需要调参。另外fitnessscore不是绝对有效的，一般工程上还会有更复杂的一些应对策略，可能还会加入速度信息，点云观测数量等等。如果没有定位结果置信度模块的话，将定位结果的置信度设置为定值那等于是一个加权平均的滤波器。总之定位结果置信度的评估还是有挺多工作可以做，但是这块我没有细致的去实现。

另外就是优化的方法，和滤波也是大差不差的，把EKF迭代起来求解，也就是IEKF，并且每次迭代都重新计算线性化点，那也就等同于是优化的方式了。但是优化的方式，实现起来会更复杂一些，我代码中的IMU和lidar的融合就是基于优化的方式，采用imu预积分，使用优化迭代求解方式得到最优解。

在我看来哪种融合方式并不是最重要的，选择合适的传感器进行融合，并且能有一个很好反馈激光雷达定位方差的模块，才是融合的关键。

你有小车的轮式里程计，如果不存在打滑的话，它的速度估计是比较准的。你使用EKF融合的话，可以使用IMU做预测，把imu的加速度，角速度都用上，然后使用轮式里程计更新速度和位姿，激光雷达也用来更新位姿。但是这样的话牵扯到IMU的初始化，要做重力对齐，还要牵扯到各个传感器标定，会比较复杂。我觉得你倒是可以尝试不使用IMU，而是使用轮式里程计做预测，然后使用激光雷达当做观测。然后使用EKF进行融合。但是前提还得是你有激光雷达的实时方差估计，不然没法应对突然的激光定位跳变。

Phymin, 2024-12-10:
目前看起来不是运动剧烈的原因，因为是小车停下来之后，定位数据发生了跳变，而小车从运动到停止也是平滑停止的，在停止之前的速度只有0.03m/s. 你说得对，我现在给激光雷达的方差是定值，所以没办法应对这种跳变。看来还是有很多工作要做

Phymin, 2024-12-12:
Hello, 想问下IVoxMap的resoution参数，会对定位波动有影响吗？为什么这个变量没有设置参数，是因为不用调整吗？谢谢 
```
void InitIVox()
    {
        IVoxMap::Options options;
        options.resolution_ = 0.5;
        options.nearby_type_ = IVoxMap::NearbyType::NEARBY18;
        ivox_map_ptr_ = std::make_shared<IVoxMap>(options);
    }
```
@zm0612

zm0612, 2024-12-12:
这个是生成体素栅格的边长大小，之所以不设置参数可调，是因为比较关键，不能随意调整，会导致系统出现coredump。这一项参数应该不会导致定位跳变的。你如果要调这个参数的话，最好把这个类的代码理清楚，然后谨慎修改。

number93soccerplayer, 2025-08-08:
> Hello, 想问下IVoxMap的resoution参数，会对定位波动有影响吗？为什么这个变量没有设置参数，是因为不用调整吗？谢谢
>
> ```
> void InitIVox()
>     {
>         IVoxMap::Options options;
>         options.resolution_ = 0.5;
>         options.nearby_type_ = [...]

你好，请问你解决了定位跳变的问题吗，我也想知道你实际测试的实际速度，0.03m/s速度太低了，你尝试过3m/s速度的表现如何吗