zm0612/funny_lidar_slam #19: 没有看到里程计结果的导出

QMonteCarlo, 2025-02-20:
能否支持把全部帧或者至少关键帧的位姿输出tum格式，看了代码，发现你的frame只是存储增量位姿，keyframe信息也很少，这样虽然轻量，但功能受限。

zm0612, 2025-02-20:
可以尝试保存keyframe的pose，使用如下TUM位姿格式保存的函数：
https://github.com/zm0612/funny_lidar_slam/blob/c103760769ee2c39d91f55c1a89f7401ca65343a/include/common/save_file.h#L12

可以尝试在如下位置，输出keyframe的pose到本地文件中：
https://github.com/zm0612/funny_lidar_slam/blob/c103760769ee2c39d91f55c1a89f7401ca65343a/src/slam/system.cpp#L743

QMonteCarlo, 2025-02-21:
只存储位置意义不大。昨天改了，保存的是关键帧，把关键帧距离调短一点也可以，还是感觉有点稀疏，不过也不错了。你的代码设计的不错，模块化做得好，表扬一下，昨天把禾赛雷达的点云结构添加进去了，能顺利跑了。

closed 2025-02-21 by zm0612

zm0612, 2025-02-21:
https://github.com/zm0612/funny_lidar_slam/blob/c103760769ee2c39d91f55c1a89f7401ca65343a/include/common/keyframe.h#L22

keyframe中也保留的有对应的点云数据，包括原始数据，去畸变数据，提取特征的数据等

zm0612, 2025-02-21:
> 能否支持把全部帧或者至少关键帧的位姿输出tum格式，看了代码，发现你的frame只是存储增量位姿，keyframe信息也很少，这样虽然轻量，但功能受限。

https://github.com/zm0612/funny_lidar_slam/blob/c103760769ee2c39d91f55c1a89f7401ca65343a/include/common/keyframe.h#L22

keyframe中也保留的有对应的点云数据，包括原始数据，去畸变数据，提取特征的数据等