zm0612/funny_lidar_slam #21: pose代表的是IMU还是Lidar？

QMonteCarlo, 2025-02-24:
你好，运行你的程序，最近导出关键帧的结果之后感觉有些困惑，发现你这里pub出来的位姿与运行的另一个纯激光雷达算法的xyz轴方向不一样，尽管大的轨迹看起来相似。所以想找你确认一下，keyframe的pose代表的是lidar还是imu？

zm0612, 2025-02-25:
你好，keyframe的pose是IMU的pose，激光雷达数据是转到IMU坐标系下的。

但是需要注意，纯激光雷达的算法，它的初始位姿是单位矩阵，也就是即使你把激光雷达头朝下启动，它也是一个单位矩阵，它和重力没有关系。

而紧耦合的激光SLAM算法，在slam系统启动时会有一步初始化的过程，它用于计算重力的方向，以保证系统启动时，坐标系统能关联上重力加速度。

QMonteCarlo, 2025-02-25:
感谢，我也是这么理解的。看看能不能在此基础上加上GPS/组合导航融合提高大范围精度，有什么建议吗？

zm0612, 2025-02-25:
你好

加gps的我一开始也有这个计划，但是太忙，一直没有时间做。

思路比较多，我也没有想好具体用什么思路。

以现在这套slam为例，我大致有两种思路：
1. 使用卡尔曼滤波的方式，将激光SLAM的结果与GPS(或者INS)的结果进行融合，这种后融合的方式会比较简单一些。
2. 使用优化的方式，构建图优化的顶点和边，将GPS当做观测，以一定的频率对整体轨迹的做优化（推荐）。

我觉得可以尝试一下第二种，可做工作会多一些，整体感觉也更加舒服。如果你也认同这个思路，期待你能帮我填坑。如果有其他思路，欢迎赐教。

祝好