zm0612/funny_lidar_slam #22: 编译缺少文件

wang119YN, 2025-02-25:
[image]，缺少安装文件，请问如何解决！

zm0612, 2025-02-25:
多执行几遍catkin_make试一下

wang119YN, 2025-02-25:
感谢您的回复，如果我想要使用mid360雷达实时建图的话，需要将livox_ros_driver改成livox_ros_driver2吗？

zm0612, 2025-02-25:
我印象中livox_ros_driver应该不支持mid 360吧！应该是使用livox_ros_driver2的驱动。

成功运行livox_ros_driver2，然后确保输出的topic有："/livox/lidar"和"/livox/imu"，然后运行如下命令：

`roslaunch funny_lidar_slam mapping_livox_avia.launch`

wang119YN, 2025-02-25:
是的，mid 360支持livox_ros_driver2的驱动，所以我在问需不需要将livox_ros_driver改成livox-ros_driver2嘛，我看你代码里面都是livox-ros_driver，还有不是使用mapping_mid360.launch建图吗？

zm0612, 2025-02-25:
我代码里面的livox_ros_driver文件夹下的内容，并不是mid 360的驱动，它只是message的格式，它并不能用来驱动Mid 360，需要你去下载livox的对应的官方驱动去运行激光雷达，然后我这边接收激光雷达的时候使用的是我livox_ros_driver文件夹下message格式去解析数据。

mapping_mid360.launch运行的是ros标准的PointCloud2格式的点云。

mapping_livox_avia.launch运行的是livox自己定义的点云格式。

wang119YN, 2025-02-25:
好的，感谢！

oldream0528, 2025-09-04:
> [image]，缺少安装文件，请问如何解决！

你好，我在编译时遇到了同样的问题，请问您是通过多次执行catkin_make解决的么