zm0612/funny_lidar_slam #30: 报错

xzh20000525, 2025-06-18:
Stack trace (most recent call last) in thread 10423:
#5    Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in ���h�
#4    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fb65c07a352, in clone
#3    Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7fb65c531608, in 
#2    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fb65c240df3, in 
#1    Object "/home/hui/funny_lidar_slam/devel/lib/libfunny_lidar_slam_lib.so", at 0x7fb65c6ab00b, in PreProcessing::Run()
#0    Object "/home/hui/funny_lidar_slam/devel/lib/libfunny_lidar_slam_lib.so", at 0x7fb65c6b0470, in IMUDataSearcher::GetDataSegment(unsigned long, unsigned long)
Segmentation fault (Address not mapped to object [0x28c])
[funny_lidar_slam-2] process has died [pid 10410, exit code -11, cmd /home/hui/funny_lidar_slam/devel/lib/funny_lidar_slam/funny_lidar_slam __name:=funny_lidar_slam __log:=/home/hui/.ros/log/975d5922-4bed-11f0-bca9-433b05dec3d1/funny_lidar_slam-2.log].
log file: /home/hui/.ros/log/975d5922-4bed-11f0-bca9-433b05dec3d1/funny_lidar_slam-2*.log

xzh20000525, 2025-06-18:
配置没问题，前面他会运行一段时间，运行的过程中，会报这个错误，然后代码会崩掉

 
检查一下imu是否有数据，以及imu topic是否正确的在config文件中配置。

zm0612, 2025-06-18:
如果是这种的情况的话，检查一下imu topic的输出频率是否稳定，imu频率建议100hz以上